steamclock/bluejay #69: Fail and handle multiple listens gracefully rather than precondition crashing

JeremyChiang, 2017-06-27:


larryonoff, 2017-12-28:
I have a lot of precondition crashes and this's really makes me thinking about another library.

JeremyChiang, 2017-12-28:
Hi @larryonoff, our office is closed until next Tuesday. Sorry we haven't been able to respond as quickly as before, but once the holidays are over we will continue maintaining and updating Bluejay actively.

I will take a look at some of these preconditions today. No immediate fixes guaranteed, but it'll be good for me to have a quick review first.

larryonoff, 2017-12-28:
@JeremyChiang thanks for the response. I think in general current queue implementation should be revised. The worst thing that it has precondition failures. I have a lot of failures in production environment. It looks that one of the case is when device disconnects and e.g. discovery operation is in progress.

Is the queue expected to be serial?

JeremyChiang, 2018-01-03:
Hi @larryonoff, I'd like to investigate these crashes you're having over the next few days with you. If you are available, we can setup a time for a voice conference as well.

To get started, I believe our apps in production don't really encounter many precondition crashes, so my initial guesses are:

**1) Usage of Bluejay:** if it's being used in ways we didn't expect, I'll need to find out how and either support it, or improve our documentation.

**2) Hardware issue:** a peripheral's Bluetooth module and behaviour are usually unique and has its own rules and restrictions for how you can interface with it, so it's always good to sanity check the hardware and make sure it is the least likely suspect for the cause of these crashes.

Moving on, here are some of the major reasons for having preconditions, and perhaps your app is encountering some of these situations.

**1) Bluejay is nil**
- This can happen easily if the Bluejay instance's life cycle is not being handled appropriately
- If a Bluejay instance is tied to a ViewController, retain cycle and unexpected deallocation are all possible depending on how the ViewController is setup and used in a workflow
- I also remember that you are using dependency injection, but I can't comment much on that because I don't really use that technique, however, make sure this is not a suspect in your app
- Basically you want to make sure you are not using Bluejay in a way that can cause Bluejay to be nil but either the Queue or the Peripheral is still in memory and being used. 
- If Bluejay is to be deallocated in specific cases, log to console and make sure that the deallocation cascades from everything from Bluejay down to its Queue and all operations and peripheral references. 

**2) No multiple listens on the same characteristic**

**3) No reading a characteristic that is currently being listened on**

**4) Queue is nil**
- This can happen in extreme rare race condition and/or threading errors, where Bluejay and its Queue are deallocated, but a queued operation is still in memory and is attempting to run and update the queue. I am not sure if this is even possible, but because so many things would have to be wrong for this to happen, the preconditions for queue != nil are sprinkled through out Bluejay.

**5) Calling regular Bluejay reads and writes and etc inside a background task.**

**6) Getting unexpected events**
- For example, if you ask Bluejay to do a disconnect, but it gets back a read response instead of a disconnection event, it would crash. My gut feeling is that this is what you might be experiencing.
- However, the queue in Bluejay is serial. An operation (scan/connect/read/write/disconnect) has to either finish successfully or fail before the next operation can execute.
- There might be a bug in Bluejay where the serial queue is not behaving as expected, but would need more info and investigation.

larryonoff, 2018-01-04:
@JeremyChiang thank you for such a great response.

Let me provide you with a bit more details.

### Intro
Please note that we don't have the final peripheral ready, so we often testing number of the peripherals. So we cannot use "persistent" connection to the device. I think this's one of the reasons of the crashes / issues.

We use a wrapper `PeripheralManager` that encapsulates all Bluetooth (via `Bluejay`) logic inside. This manager is initialised when the app starts and this manager behaves like a singleton. So there's only one `Bluejay` instance.

We have two screens that interact with `PeripheralManager`
1. Devices picker screen. Only scans devices.
2. Device screen. Connects and communicates with device. It's important to note that app calls disconnect when this screen is closed (on deinit).

Crashes usually happen on devices picker screen. I've checked two crash logs that we've observed and it's when scan action is in progress and disconnect happened. See two dumps of crashes dumps enclosed to the archive [crashes.zip](https://github.com/steamclock/bluejay/files/1603042/crashes.zip).

### Quick-fix
I've found a way to reduce (maybe even avoid crashes), it's
```
        guard
            !peripheralManager.isConnecting &&
            !peripheralManager.isDisconnecting
        else { return }
```

This's the solution that I don't like. Since I cannot queue scan while `isDisconnecting == true`.

### Possible TODOs
I have the following ideas on improving `Bluejay`.
1. Implement additional preprocessing of Bluetooth events (e.g. disconnect events) before processing them by a concrete action.
2. Replace `preconditionFailure` with `assertionFailure`. Probably in a case of unexpected events just cancel actions, but not all (e.g. don't cancel scan).

### PS
I agree that the library is used not exactly as it's designed, i.e. it's intended to be used with one peripheral. So we had to adjust our logic to this. But nevertheless we have often disconnects / connects / scans. I believe that adding multiple peripheral support would be the next big step for Bluejay.

### Call
I'm in GMT+3 (Minsk) time zone, according to your profile you're in PST time zone. It's -11 hours. I can do from my 8 AM (9 PM yours) till my 10 PM (it's 11 AM yours).

JeremyChiang, 2018-01-04:
Hi @larryonoff, thanks for the context and information. They are very helpful.

From the crash log, it seems like a disconnection event is happening while a **scan** operation is at the top of the queue.

I believe this is typically not possible in a normal use case and with only one peripheral, as we've built in protection against starting a scan while Bluejay is connecting or already connected. See line 88 in `Queue.swift`. In other words, under normal usage of Bluejay, we should never encounter a disconnection event while a scan is still at the top of the queue.

```
else if queueable is Connection {
    // Fail the connection request immediately if Bluejay is still connecting or connected.
    if bluejay.isConnecting || bluejay.isConnected {
        queueable.fail(BluejayError.multipleConnectNotSupported)
        return
        }
            
    // Stop scanning when a connection is enqueued while a scan is still active, so that the queue can pop the scan task and proceed to the connection task without requiring the caller to explicitly stop the scan before making the connection request.
    if isScanning {
        stopScanning()
        return
    }
}
```

I would take a look at your custom logic, and investigate why you are receiving disconnection events from devices when the user is looking at the "Device List Screen". Even if you need to work with multiple devices, when the "Device List Screen" is active and before it tries to scan, it should probably make sure all devices are disconnected before scanning anyway.

If that is not possible for business reasons, and you need to connect or disconnect devices and scan at the same time, then a better temporary workaround for your fork before we add official multi-device support probably looks like this:

1) Modify the above code in `Queue.swift` starting at line 88 so that Bluejay is able to start a scan even when connecting or connected

2) Modify the `process` function in `Scan.swift` starting at line 100, so that the scan operation is able to process connection and disconnection events as well – it should probably just recognize those events but probably doesn't need to do anything. This should eliminate the precondition crashes you're running into.

The caveat here is you still don't want to be reading or writing or listening in your app while scanning. Otherwise you need to make even more modifications to the `process` function in `Scan.swift`, and it would get gnarly quick.

Give this a try and keep me posted. Thanks!

JeremyChiang, 2018-12-03:
Closing for inactivity.

closed 2018-12-03 by JeremyChiang