steamclock/bluejay #74: How to discover unknown services of a peripheral?

haemi, 2017-07-05:
I would like to let the user decide which peripheral to connect to, how can I discover services and characteristics then?

apike, 2017-07-06:
@haemi I believe if you do a `scan` but you pass in `nil` for the `serviceIdentifiers`, you should get callbacks for all visible bluetooth LE devices, which you can then filter and present to the user.

In terms of iterating the services and characteristics available on a peripheral, I believe Bluejay supports this but I can't see how to do so from the documentation. @JeremyChiang, can we point the way and then turn this into an issue for documenting the method(s) for doing this?

JeremyChiang, 2017-07-11:
While Bluejay is intended to be used in situations where you have a specific peripheral that you have documentation for or knowledge of, I don't see why we can't add API's to allow exploration of services and characteristics.

Because this is not a difficult feature and is nice to have, I will put this in the backlog and I hope we can get to this within a few weeks at most.

apike, 2017-07-12:
Let's split this into two issues: in the short term, we should make the docs clear that Bluejay is optimized for quickly connecting to known devices that you have some sort of documentation for. Then we can have a longer term issue for exploring what we'd need to add to support exploratory open-ended apps or hack projects against hardware without documentation.

JohnyKumar27, 2018-08-03:
Hi Jeremy, When can we have this feature of loading all services and characters?

JeremyChiang, 2018-08-03:
Hi @JohnyKumar27 sorry about not getting to this feature due to other issues and projects taking over my time. I will push this up the priority list, but we may not be able get to this feature right away. I can't give you an ETA right now, but we're not abandoning this feature either.

Meanwhile if you are able to come up with a solution, we could use a PR ;)

I will try to address this as soon as I can once I am able to do another round of full-time development for Bluejay in the next few weeks.