steamclock/bluejay #88: Support iOS 9.0

MenesidisMich, 2017-07-28:
I would like to use the library in iOS 9.0 and later. How would you recommend to replace the "Dispatch.dispatchPrecondition(condition:" in order to be compatible with iOS 9.0 ?

apike, 2017-08-02:
@MenesidisMich Good question! @nbrooke is currently integrating Bluejay into an iOS 9 app so we probably do need to add this.

JeremyChiang, 2017-08-15:
Bluejay version 0.1.2 now compiles with iOS 9.3, but loses the thread checking safety when using background task. It is not an essential feature to have, and it's also not a high priority to add for 9.3 at this moment. For now, the attitude towards supporting for iOS 9.3 is probably more about temporary convenience, rather than a full intention to have complete backward compatibility.

With iOS 11 coming out soon and iOS 9 market share expected to dip even more, we may yet again drop iOS 9 support before we even get a chance to fully support iOS 9, if that makes sense :P

JeremyChiang, 2017-10-12:
@apike, @nbrooke should we still try to support iOS 9? According to https://david-smith.org/iosversionstats/, iOS 9 is just shy of 10%, and would probably decrease even more as Christmas and iPhone X come around.

apike, 2017-10-12:
Depends how much work is needed in my mind - if it's simple then let's do it, if it requires substantial changes or would make usage on iOS 10 or 11 worse, then probably not.

nbrooke, 2017-10-13:
We have at least one client project still using iOS 9 + Bluejay. So I don't think we can just straight up ditch it yet. I'm pushing for the next major release to be iOS 10 only, so if that goes through we should be okay to drop it. And we definitely shouldn't put much into supporting it better (i.e. trying to fix the thread safety stuff to work on iOS 9)

JeremyChiang, 2018-08-24:
Let me know if it is **not** a good idea to drop support for iOS 9 when iOS 12 is released later this September or October. We can also drop iOS 9 next January instead to give people (ourselves included) more time for transition.

nbrooke, 2018-08-27:
We are now iOS 10+ only for the project the I was unsure about above, so I don't think there are any other internal roadblocks to dropping it now. 

I think the only open question is: how should "tie it off" for the benefit of any external users. Like should we just say drop it now and say 0.6.4 (or whatever the most recent version is when we actually do it) is the last iOS 9 supporting release, or do we try and hit some reasonable major milestone first (fix the couple outstanding crash bugs and declare it 1.0 with iOS 9 support, then start working on a 1.1 or 2.0 that drops iOS 9 support and has some of the larger new features like Codable support or whatever).

JeremyChiang, 2019-01-11:
Implemented.

closed 2019-01-11 by JeremyChiang