steamclock/bluejay #91: Why Receivable not failable or throwable?

larryonoff, 2017-09-04:
I like the idea of this library. I haven't tried it yet. But it looks that it really simplify CoreBluetooth API.

I'm curious why `Receivable` `init(bluetoothData: Data)` doesn't throw or can return nil? What to do if there's invalid data from Peripheral?

apike, 2017-09-06:
@larryonoff Good question! Probably @nbrooke will have a good answer for you.

nbrooke, 2017-09-06:
There isn't a really good reason other than we haven't needed it yet and it never occurred to anyone 😄 

Lots of the data being pulled off devices isn't complex enough for there to be meaningful deserialization failures (if you ask for an Int32, there's no value the device could send you that would really be "invalid"), and it just happens for the few more complex things that we have that implement Receivable also don't have parsing failures like that (for example, we have an event record we pull off the device for one project that could have its type field validated, but our approach is actually to just store the invalid record anyway, on the theory that it could be that a firmware upgrade has added a new type we don't understand, and so it's better to store it but ignore it than discarding it, and even if not keeping that in the DB is useful for debugging whatever issue with the firmware cause the invalid data to get sent).

That said, I think that Larry is probably correct that in general Receivable SHOULD have some way to represent a parsing failure, because it's definitely something that can happen.

I think having Receivable init throwing is a better solution than returning nil, because if there is a parse failure we do want to be able to send back an error describing the specific failure, rather than just knowing parsing failed.

We should probably go ahead and do that relatively soon, since it will be an API breaking change, even though it's a pretty trivial one to make / fix.

nbrooke, 2017-09-06:
cc @JeremyChiang

larryonoff, 2017-09-06:
I have a real case why Receivable protocol update is needed. A big pack of data is received that doesn't fit in one packet. This data is represented by structure with the fields like: timestamp, identifier, array of doubles.

The library concepts are really awesome. So to make the library popular It should cover a lot of developer needs.

PS. I think the best approach will be something like Codable / Encodable from Swift 4. But this obviously more complicated solution than just adding throwable init.

JeremyChiang, 2017-09-06:
I should have some time to add a throwable init this week. ETA around Friday. Will need to bump version and update clients' projects as well.

apike, 2017-09-06:
Since this is breaking it would need to go to v0.2 I believe.

nbrooke, 2017-09-07:
Yup. 

We should also consider using Codable at some point. I do have some concerns about that, mainly that Codable is definitely oriented very strongly around named fields as opposed to ordered fields which is what you really need for this, so you might end up either with having to write a lot of the encode/decode methods manually anyway to enforce specific orders or byte widths, but it's definitely worth doing some experimenting with. I'll file a separate bug for that.

larryonoff, 2017-09-07:
@nbrooke I didn't mean using `Codable` 1-by-1. I meant implementing something similar like `Encoder / Decoder`. `Codable` doesn't work for the `Data` case.

I think it should be something like
```
struct SuperPacket: BluetoothCodable {
     let doubleValue: Double
     let int16Value: Int16
     let doublesArray: [Double]

     init(_ decoder: Decoder) throws {
          doubleValue = try decoder.read()
          int16Value = try decoder.read()
          doublesArray = try decoder.read()
     }
}
```

Looks this case looks very trivial. What about keyPaths from Swift 4?

nbrooke, 2017-09-07:
We should move discussion of the codable stuff to the bug for that purpose: #92

JeremyChiang, 2017-09-11:
@larryonoff Let me know if this is what you had in mind: https://github.com/steamclock/bluejay/pull/93

Will be happy to add or fix anything I've missed before bringing it to master.

larryonoff, 2017-09-12:
@JeremyChiang responded in #93

closed 2017-09-25 by larryonoff