steamclock/bluejay #94: Suggestion to add SwiftLint

larryonoff, 2017-09-12:
I have a suggestion on adding swiftlint into the project. I can create PR if community doesn't have any doubts about this.

Please feel free on changing configuration that I provided below.

```
included:
  - TBD
excluded:
  - Carthage
  - Pods

disabled_rules:
  - file_length
  - force_cast
  - function_body_length
  - function_parameter_count
  - large_tuple
  - nesting
  - redundant_optional_initialization
  - type_body_length
  - type_name
  - variable_name
  - weak_delegate

closing_brace: error
comma: error
colon: error
empty_count: warning
function_body_length:
  warning: 100
  error: 150
opening_brace: error
return_arrow_whitespace: error
statement_position: error
todo: warning
trailing_semicolon: error

```

JeremyChiang, 2017-09-13:
I've not used SwiftLint before, but realm seems to have a great and smart team. I love their NSPredicate cheat sheet. If this is one of the best linters out there for Swift and is used by other major libraries, I'm down to adopt this.

larryonoff, 2017-09-14:
@JeremyChiang I think `SwiftLint` is the most mature linter for Swift. There's not official linter from Swift team, but Realm did a very good job by implementing open-source `SwiftLint`. Other iOS devs that I know use `SwiftLint`.

I'll create PR over the week-end.

larryonoff, 2017-09-14:
example of framework that uses SwiftLint: https://github.com/Moya/Moya

larryonoff, 2017-10-13:
Hopefully will start working on this issue these weekend.

PR #180 (2018-12-06): v0.7.1 - Add SwiftLint
### Fixes
- Closes #94

### Summary of Problem:
- Code was not pretty enough

### Proposed Solution:
- Add SwiftLint

### Testing Completed and Required:
- Compile
- Smoke test using a few existing projects that use Bluejay heavily
closed 2018-12-06 by JeremyChiang