steamclock/bluejay #98: Bluejay.connect doesn't call completion on iOS 11

larryonoff, 2017-09-20:
I have analyse the source code. As result I've found the following:

It's important to note that restoration is set to disabled.

1. I use very simple dependency injection approach, so that some dependencies are created in `AppDelegate.init`.
2. `Bluejay.init` is called.
3. init executed the following code `shouldRestoreState = UIApplication.shared.applicationState == .background` as result equals to `true`.
4. At some time I call `Bluejay.connect`. But it just skips real connect since of shouldRestoreState is `true`.
```
        if (shouldRestoreState) {
            // Cache requested connect, in case restore messes up unexpectedly.
            peripheralIdentifierToRestore = peripheralIdentifier
            return
        }
```

As a quick fix I method `Bluejay.start` I set shouldRestoreState to false if restoreMode is `disabled`.

As I understand `centralManager(_:willRestoreState:)` not called. Since I have restoreMode disabled.

Could anyone please comment help with issue?

Thank you!

larryonoff, 2017-09-20:
What about `Blujay.init` with parameters?

JeremyChiang, 2017-10-11:
@nbrooke should `Bluejay.start` check for uncanceled background task and cancel it if it is still running by the time `start` is about to be executed? I think that will be the case if Bluejay is initialized so early and the app is not even active yet.

In the cases of actual background restoration where Bluejay is re-initialized while the app is in the background, I don't think `start` will be called most of the time since its scope should be closely tied to view controllers, but the fact we can't guarantee or enforce it also bothers me – so I'm not sure how best to avoid cancelling a background task accidentally.

nbrooke, 2017-10-12:
Yeah, it probably should cancel the background task and set set `shouldRestoreState` to false in `start` if you aren't actually doing a restore. That's definitely wrong given the approach we are taking now, IMO.

I've never been super happy with that background check and `shouldRestoreState thing` in general, it's a little too magical. It might be better to take a different approach to triggering that. Even if it's a little more complicated for the end users, state restoration is a pretty complicated thing to implement anyway, so making them (for example) call a static function that must be called from `application:didFinishLaunchingWithOptions` is not super burdensome. I'll file a bug for that.

All that said, it's quite surprising that during normal app launch the applicationState is `.background` when the `init` on AppDelegate is called (I believe it would be `.active' by the time you hit `didFinishLaunchingWithOptions`). Just checked it, and that does seem to be the case though.

@larryonoff: I'm a little suspicious about having code called from `init` in a UIApplicationDelegate subclass. That is called dangerously early in initialization, I wouldn't trust that 100% of code can be safely run there. The rule of thumb I've always followed is that even though there are ways to get your code to run before the start of `application:didFinishLaunchingWithOptions`, they should be ignored, and you shouldn't try to run code any earlier than that.

larryonoff, 2017-10-12:
@nbrooke I also don't like the approach on doing something in `AppDelegate.init`. I wanted to initialise AppDependencies there to make them under `let`. But I'm migrating already to `application:didFinishLaunchingWithOptions`.

> All that said, it's quite surprising that during normal app launch the applicationState is .background when the init on AppDelegate is called (I believe it would be .active' by the time you hitdidFinishLaunchingWithOptions`). Just checked it, and that does seem to be the case though.

This's a case for iOS 11 and may be bug.

closed 2017-11-11 by larryonoff