steamclock/bluejay #102: Figure out a better way to do the start of background state restoration

nbrooke, 2017-10-12:
The quirks of background Bluetooth restoration require that we do some stuff (in particular, start a background task so that we won't get immediately stopped again by the OS and quickly restart the peripheral managers) during `application:didFinishLaunchingWithOptions:`

Right now we do that with a bit of a hack by just assuming that the Bluejay object will be initialized at the right point that we can detect that we are starting in the background and early enough that starting the background task will happen before we are shut down, but those assumptions are only good for a subset of the possible ways you can initialize Bluejay (and indeed caused a bug even without state restoration in #98)

I think we need to try and do a better job of figuring out when we are allocating a Bluejay instance that is eligible for state restoration, probably by making some effort to force certain calls to take place during didFinishLaunchingWithOptions, and not allowing allocation of Bluejay instances with state restoration at other times.

JeremyChiang, 2018-10-23:
`hasStarted` boolean is already implemented. The improvement to `isBluetoothAvailable` is now also implemented in PR #175

JeremyChiang, 2018-12-04:
Need to test this with the app killed from Xcode and remain in the background. Try to trigger a connection or sync from the peripheral without bringing the app to the foreground.

JeremyChiang, 2019-01-11:
Implemented.

closed 2019-01-11 by JeremyChiang