steamclock/bluejay #110: "Sensor Location" list does not work

abey, 2017-11-07:
### Current Behaviour
Users are not able to select a specific location from the list as tapping any of them does nothing and it's value always stays at Unknown.

### Expected Behaviour
A user should be able to select a specific location from the list and it should update the value of Sensor Location to the chosen item.

### Steps to recreate the bug
1. Open BlueJayDemo app and connect to a bluetooth peripheral (for this example, I created a Heart Sensor using Lightblue)
2. Click the name of the device to open the Heart Sensor page
3. Click **Sensor Location** and choose any location from the list

### Screen Recording
[image]

JeremyChiang, 2018-12-03:
Added temporary solution here: a5798d66809269f4481734f08c2bc45d0e2dbe86

You need to enable write permission for the Body Sensor Location characteristic in LightBlue, but that is not obvious at all. Added some instructions in that screen to temporarily address this issue.

In the future we won't be using LightBlue in our demo anymore, so this should be sufficient for now.

JeremyChiang, 2019-01-11:
No longer relevant in the latest version as the demo app has changed and doesn't use this feature anymore.

closed 2019-01-11 by JeremyChiang