steamclock/bluejay #112: Scan duration/timeout is not supported

avtr, 2017-11-08:


mdxs, 2017-11-12:
I'm actually wondering how to implement this now; as I've making a simple App which should scan for some devices for a while (so I want to stop/cancel scanning after a number of seconds). Allowing some interaction on devices (while scanning and/or after the scan). Actually a bit like the nRF Connect (from Nordic Semiconductor) is doing, which allows the user to set a scanner Timeout; and provides in top left corner a "Scan"/"Stop Scanning" button to toggle as well.

Should I try to hold on to the queued Scanner object in order to send a cancel/stop command?

larryonoff, 2017-12-23:
Implemented in PR #125

JeremyChiang, 2018-01-03:
Implemented by @larryonoff, thank you :)

closed 2018-01-03 by JeremyChiang