steamclock/bluejay #113: ScanDiscovery shouldn't expose CBPeripheral, should expose a PeripheralIdentifier

nbrooke, 2017-11-08:
Right now the ScanDiscovery contains a raw CBPeripheral, I think that's with #103 changed it's the only case where real CoreBluetooth objects leak into the public interface, so we should probably remove it on general principle. 

This also a bit of a usability issue there, that one thing you might want to do with the scan discovery is to store the peripheral identifier, but because you only have a CBPeripheral object, you have to manually wrap it into a Bluejay.PeripheralIdentifier to use it with the rest of the system.

I'd suggest removing the CBPeripheral from ScanDiscovery and replacing it with a name String and a PeripheralIdentifier. I don't think there is any other properties of the real CBPeripheral that you can usefully get from it that you might want to act on, but if so we should probably add them as well (canSendWriteWithoutResponse, maybe? Though I'm not sure what a caller could do with that exactly.) Even if there WAS a good reason to keep the real peripheral around, we definitely hold have an easy way to get a Bluejay.PeripheralIdentifier

larryonoff, 2017-11-09:
I think it's a right decision in terms of the framework. So I've created PR #117.

larryonoff, 2018-01-04:
this's already done on #117

closed 2018-01-06 by nbrooke