steamclock/bluejay #114: PeripheralIdentifier (maybe others) should conform to Equatable and Hashable

nbrooke, 2017-11-08:
Right now you can't compare PeripheralIdentifiers or store them in sets, both of which are useful in some situations.

It's trivial to add retroactive conformance yourself (I did when I had the problem), but it would be nice to just support that out of the box.

ServiceIdentifier and CharacteristicIdentifier would probably make sense to conform to these as well, though the cases you'd need it are somewhat less obvious than peripheral.

Here's the code I used to implement it on PeripheralIdentifier:

```
extension PeripheralIdentifier: Equatable {
    public static func == (lhs: PeripheralIdentifier, rhs: PeripheralIdentifier) -> Bool {
        return lhs.uuid == rhs.uuid
    }
}

extension PeripheralIdentifier: Hashable {
    public var hashValue: Int {
        return uuid.hashValue
    }
}
```

larryonoff, 2017-11-09:
Plus #118.

larryonoff, 2017-12-23:
@JeremyChiang @nbrooke please close this issue. It's already implemented and merged in #117 and and #118

closed 2017-12-23 by JeremyChiang