steamclock/bluejay #116: Bluejay.isBluetoothAvailable should work prior to BlueJay.start

nbrooke, 2017-11-08:
Right now it traps prior to start, like many other methods, but it seems like in this case there is a clear answer if Bluejay isn't started (i.e. false), and being able to check that prior to start could simplify some code that depends on asynchronous events coming in that might need to touch the bluetooth state, i.e. you can put in a few judicious `guard bluejay.isBluetoothAvailable else { return }` and handle a lot of cases where bluejay either hasn't started or hasn't finished initialization yet

Another option would be to add an "isStarted" property for checking that, which would give you even more flexibility, but I think that's probable unnecessary in practice.

JeremyChiang, 2018-10-23:
`hasStarted` boolean was already implemented. The improvement to `isBluetoothAvailable` is now also implemented in PR #175

JeremyChiang, 2018-11-02:
Implemented.

commit c7e10437 (2018-11-02): Allow checking is bluetooth available before starting Bluejay, closes #116 (#175)
commit 3c9156a1 (2018-12-06): v0.7.0 - Improvements to initialization, disconnection, and callback structures (#177)

* Allow setting CBCentralManager initialization flags (#165)

* Add the option to change the standardConnectOptions used for the CBCentralManager from outside of the Bluejay module

* Scope the connection options for the CBCentralManager to the connections (so each Bluejay instance has its own connection options)

* Filename should match the enum name

* Simplify interface for specifying manager and peripheral options

* Further simplify connection options

* Improve unwrapping of the connection options opt [... cut]
closed 2018-11-02 by JeremyChiang