steamclock/bluejay #120: Could indicate background restore with connect has happened better

nbrooke, 2017-11-14:
Right now, Bluejay doesn't give any indication that a background restore has happened in the form of a callback or anything if the background restoration is due to a connect rather than a listen (listens do get callbacks, obviously).

Having a clear signal about that isn't technically necessary, it's not impossible to figure out what has happened from available clues (you can check the restore state or app background mode yourself in didFinishLaunching, you know that restoration always happens by the time bluetoothAvailible is called so you can figure things out there by checking connection state, etc.), but doing it right now does involve some knowledge of how those things work and what order things happen in that is non-obvious and may involve the app tracking some state that mirrors Bluejay internal state that we could expose.

It would be nice if there was some clear point where you could say something like "if didRestore { doStuff() }" with minimal fuss. Since you usually can't take any action until the bluetooth hardware is initialized, you generally need to do that work in the bluetoothAvailable callback anyway, so I think the right approach is to have that callback take another parameter that indicates that a restore just finished, though there definitely are other designs that might make sense (a separate callback on ConnectionObserver, or a flag on the Bluejay object itself).

Even though this could be done independently, it probably makes sense to do this at the same time as #102.

JeremyChiang, 2018-12-03:
Similar to #156, this is also better handled now: https://github.com/steamclock/bluejay/commit/bb3763c566c797424dd56977e7a47b4e6a63a107

1. The `BackgroundRestorer` protocol now gives you a clear indication of whether there was a restoration attempt. If there was a restoration attempt, it will definitely either successfully restore, or fail with a reason and error.
2. You are also required to return a callback in which you can do the stuff needed after a restoration success or failure
3. Bluetooth available is not broadcasted at the end of a successful or failed restoration event to separate `ConnectionObserver` from `BackgroundRestorer`
  - Events for `ConnectionObserver` are for things that happen during normal use
  - Events for `BackgroundRestorer` are unique to restoration and also unique enough to not have any overlap with `ConnectionObserver`

JeremyChiang, 2019-01-11:
Implemented.

closed 2019-01-11 by JeremyChiang