steamclock/bluejay #134: SIGABRT error when creating an instance of bluejay (private let bluejay = Bluejay()).

maniacaris, 2018-02-10:
I am trying to write an app that is an adaptation of the heart rate demo. When I click the button to segue to the connection screen, the app crashes with a SIGABRT error when I create an instance of bluejay. I apologize if this issue isn't necessarily related to your framework, I'm new to XCode so my knowledge regarding these things is fairly limited.
[image]

nbrooke, 2018-02-13:
That definitely should work in general and does fine in other places, we would need a little more detailed crash information than that to debug if it is a Bluejay problem. There should be more info printed out about what is causing the SIGABRT in the output window. You should also check if it's actually a problem with the scanButton not been wired up in your storyboard. That could cause this sort of crash (due to it being an IUO), and maybe it is just showing up on the wrong line.

maniacaris, 2018-02-14:
Thanks for the reply. Ended up completely uninstalling and reinstalling the pod from the project and it works now without errors.

closed 2018-02-14 by maniacaris

shibbs, 2021-11-08:
FWIW I had the same issue and it was due to a missing "NSBluetoothAlwaysUsageDescription" I added that into the info.plist and then it worked