steamclock/bluejay #135: Connection State Function Usage

maniacaris, 2018-02-26:
I am either not using the bluejay.isConnected function correctly or it is not working as I understand it is supposed to. I am simply trying to add a connection status label at the bottom of my app's home screen that displays the status of the connection and the device it is connected to. I have gotten this to sort of work using a global variable which obviously isn't ideal. 
1) What does this function actually return?
2) Do you need to generate a connection observer in every view controller that utilizes the bluetooth connection?

Note: The debugger code is not printing the text either so I have a feeling it is not entirely due to any coding errors in regards to updating the UI. 

[image]

JeremyChiang, 2018-02-26:
Hi, thank you for your question. The `isConnected` variable is meant for a one-time use and evaluation of the connection status at the time of accessing that boolean.

In your code example, I see no connection calls between `bluejay.start` and `if bluejay.isConnected`. So what you are doing is essentially this:

1. Initialize Bluejay
2. No Bluetooth operation is performed
3. Checks whether Bluejay is connected, and of course nothing is connected because you've not made any connections, so your line 25 will always return false.

If you want to update your UI, what you need is a callback on current and future state changes. So yes, you will need to setup a connection observer, and that is the proper way get your UI to receive notifications on connection state changes and to update accordingly.

JeremyChiang, 2018-02-26:
This is not a bug in Bluejay but rather an API question, so I will close this for now. I'd be happy to continue answering any questions.

closed 2018-02-26 by JeremyChiang

maniacaris, 2018-02-27:
Thank you for the quick response I appreciate it.