steamclock/bluejay #144: didModifyServices delegate is not implemented

shekhar-suman, 2018-03-21:
`- (void)peripheral:(CBPeripheral *)peripheral didModifyServices:(NSArray<CBService *> *)invalidatedServices`

This delegate implementation is missing from Peripheral class. Hence if Peripheral goes down or shut down its service or start its service, no bluejay object delegate is called. 

Hence if a listener is attached to peripheral's characteristics and service goes down then listener is not getting reattached automatically.

Danappelxx, 2018-08-22:
For such a polished library this seems like a pretty major oversight. Any progress? If not I'll take a jab myself.

apike, 2018-08-23:
It looks like @shekhar-suman had a patch for this and it just slipped through the cracks – @JeremyChiang did you see this? https://github.com/shekhar-suman/bluejay/commit/aa492da70f37ed567592ce34e05da12853e407ff

JeremyChiang, 2019-02-05:
I would like to add some notes after having spent time looking into this in detail:

### Summary
A complete support for how this should actually work is pretty complicated, so rather than doing everything in one go, we can release partial support for this in phases with documentation explaining the current can-dos and can't-dos.

### Why full support is complicated:
@shekhar-suman touched on a very important point: when a service goes down, Bluejay as a smart and easy-to-use library should not only notify, but also probably automatically fail any existing listens associated with invalidated services.

The notify part is easy, and @shekhar-suman has already done that in his fork, though I wouldn't use `ConnectionObserver` for that, and I will show and explain why in my upcoming PR.

The failing of existing listens part is slightly trickier, but also not the hardest part. 

Where it really gets complicated is `didModifyServices` is also called when new services are added, except this delegation won't tell you which services are added, because the only parameter it has is for invalidated services 😫

```
optional func peripheral(_ peripheral: CBPeripheral, 
       didModifyServices invalidatedServices: [CBService])

This method is invoked whenever one or more services of a peripheral have changed. A peripheral’s services have changed if:
  - A service is removed from the peripheral’s database
  - A new service is added to the peripheral’s database
  - A service that was previously removed from the peripheral’s database is readded to the database at a different location
```
[Apple doc](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/1518865-peripheral) specifically tells us to use `discoverServices` to find out which services are (or might have been) added.

What all of this means is: `didModifyServices` is not just used for figuring out when services are down, but also when they are added or revived, and Bluejay "should" make all of that possible and easier...? A question mark because maybe this is too advanced already, so as long as Bluejay just gives you a way to get started, that **could** be enough and justified.

The complete support may result in the following additional requirements:
- Add an API for just discovering services, so that users can figure out the latest available set of services, as being notified of services changes but not being able to evaluate services is kind of pointless in many situations
- If Bluejay needs to be able to easily tell us that a new service is available, that means we also need to start tracking and caching the discovered services to use it as a reference through out the session

There might be more, but moving on...

### Proposed Plan
1. I think phase one we should just make the delegation and event available, so that users can at least handle the "services down" situation, and manually tear down any listens, Bluejay, and UI logic.
2. Phase two we should add a discover services API to partially support the "new services available" scenario, but the user will still have to guess or be aware or always assume that new services are potentially added when `didModifyServices` is called.
3. Phase three, maybe add the auto fail existing listens part when Bluejay is confident that the services with listens are down.

The first 2 phases are fairly easy and incremental changes that shouldn't cause any regressions. And after that, we might need to re-evaluate again on whether it's really worth digging into this further and add all of those smarts I've mentioned, as doing everything will require making architectural changes to the queue and introduce more state tracking - therefore possible bugs.

apike, 2019-02-21:
Let's go with 1 (make the delegation and event available and add a documentation note that if you want to properly handle a new service appearing, a lot of manual work is necessary) and file 2/3 as a future issue.

PR #193 (2019-03-21): v0.8.2 - Add initial support for did modify services
Fixes #144

Filed #194 for future work.

### Summary of Problem:
No way to react to a connected peripheral removing or adding services as it's being used.

### Proposed Solution:
Expose the `didModifyServices` delegation and allow Bluejay apps to monitor that callback.

### Testing Completed and Required:
- You can now add and remove the chirp service in **DittojayHeartSensorDemo**
- You can then test and verify the adding and removing of the chirp service using **BluejayHeartSensorDemo**
- If the chirp service is removed, BluejayDemo should [... cut]
closed 2019-03-21 by JeremyChiang