steamclock/bluejay #148: Disable auto connect once connection has been lost

DrAma999, 2018-04-04:
During some test we found out that after a connection has been lost, bluejay tries to connect again to the device.
While this is an appreciable feature, there is no way to disable it.
In some case, for security reasons, after the connection is mandatory to write a signature or a hash in a specific characteristic. Usually if you don't the connection drop after few seconds, or drops if we try to write to a characteristic before sending this "signature".
In this case Bluejay continuously try to connect to the peripheral, resulting in a loop of continuous disconnection and connection.
I have two solution for this:

1. Modify the source and disable this behavior
2. Use a connection observer, once the connection has been made to the peripheral, in the observer call back I write the signature to the characteristic.

In the latter I have one doubt: If I do that can I be sure that the signature writing it will be processed before everything in the queue?
To make an example, my operations will be:
->Connect to the peripheral
->Write a command
->Listen in notify
Can I be sure that the connection observer callback (that will contain the write of the signature) will be executed before the write of the command and the notify?
Thank you,
Andrea

JeremyChiang, 2018-04-09:
Hi Andrea, thank you for your feedback. We already have a connection observer callback you can use. See https://github.com/steamclock/bluejay#bluetooth-events

I believe you can add your security-related writes in that connection callback. As to whether your security-related writes are first in the queue, they should be, as long you setup your observers properly in your application code.

If you have observers in multiple views and controllers, you might need to coalesce the connection and disconnection events into a single and global flag that your application keeps track of. Then when you are sure you are getting a new connection callback where you need to write some initialization or security commands, you can do so there.

We'll look into adding support for turning that off in the near future, but meanwhile see if there's way to tell whether authentication is successful before the device drops you, and manually call disconnect if the security commands fail. Manually disconnecting the device while it is still connected will turn off auto-reconnect until the next successful connection.

DrAma999, 2018-04-10:
Super.. thank you!

JeremyChiang, 2018-07-26:
I will be releasing a new version of Bluejay later today, and it will give you more options to deal with auto-reconnect, so I'm going to close this. Thanks for the feedback!

closed 2018-07-26 by JeremyChiang