steamclock/bluejay #150: Is this line not supposed to be as follows?

ngenworx, 2018-04-10:
https://github.com/steamclock/bluejay/blob/c1b0a4f2dc60ad92e13ef9a3bfa94c9c72b9c5c9/Bluejay/Bluejay/Queue.swift#L171

```
else if !blukee.isConnected && ((queueable is Scan) || (queueable is Connection)) {
           // Fail any queueable that is not a Scan nor a Connection if no peripheral is connected.
           queueable.fail(BlukeeError.notConnected)
        }
```

The code does not match the comment.
Also if multiple connection requests fail the queue will build up.
Am I missing something?

JeremyChiang, 2018-04-10:
The code you posted looks like a fork of Bluejay with some changes made, and it doesn't actually match the original source code.

`else if !bluejay.isConnected && !(queueable is Scan) && !(queueable is Connection) {`
vs
`else if !blukee.isConnected && ((queueable is Scan) || (queueable is Connection)) {`

Note the if-statement of "blukee" is missing the exclamation marks.

ngenworx, 2018-04-10:
Ahh... ok... makes sense now

closed 2018-04-10 by ngenworx

apike, 2018-04-10:
@ngenworx As an aside, we're always curious how folks are benefitting from Bluejay, so it'd be interesting to hear how you learned about this "Blukee" fork, and if it has any modifications from the standard open-source Bluejay worth knowing about.

ngenworx, 2018-04-10:
I think I copied a version from @DrAma999, so that was me playing around with the code. I looked at all the BlueTooth SDKs available and liked Bluejay's the most, but have not committed to which to use yet.