steamclock/bluejay #161: Crash report: DiscoverService.swift:51

FIndustries, 2018-08-11:
Many users have this problem.
This bug is present in every version of ios (from ios 9 to ios 12).

I still do not understand what cause this crash but it happens when the `disconnected(from peripheral: Peripheral)` and `connected(to peripheral: Peripheral)` functions are called.

This is my log before the crash
```
...
23 | \| | 09:38:29:183 (UTC) | \| | disconnected(from peripheral: Peripheral) 
24 | \| | 09:38:36:286 (UTC) | \| | disconnected(from peripheral: Peripheral)
25 | \| | 09:38:58:320 (UTC) | \| | connected(to peripheral: Peripheral)
--CRASH--
```

[issue.txt](https://github.com/steamclock/bluejay/files/2280132/issue.txt)

JeremyChiang, 2018-08-14:
If possible, seeing how you're using the Bluejay API in your app would also help. For example, could you show us 15 ~ 30 lines of code that surround the problem area?

From the crash log, however, I can already see your Bluejay got into a bad state where the top of queue should have been a `Connection` operation for handling the `didConnect` event, but instead, there's a `DiscoverService` operation instead of `Connection`.

It's unclear at this point whether that's caused by a bug or a lack of safety check in Bluejay (there already is some), or a misuse of the Bluejay API - in which we could potentially add further protection and documentation to prevent people from trying to do the same thing again in the future. Thanks!

JeremyChiang, 2018-08-22:
@Danappelxx thank you for your explanation. Whenever you want to perform more than one Bluetooth operations in a synchronous order, you should use the `bluejay.run(backgroundTask:` API. See https://github.com/steamclock/bluejay#batch-operations for more info.

In your code sample, what you are actually telling Bluejay to do is to start the listen first, and then start the read immediately after the listen has started, **regardless** of whether the listen is done and ended. So yes, that easily creates an undesirable race condition.

Standard Bluejay API calls are all asynchronous and more meant for convenient and quick one-off/one-time convenient use where the timing of the results do not matter. For example, after connecting and setting up, you may have a button that reads a value and displays it in the UI - then you'd just use a regular Bluejay API. 

In all other non-trivial cases, especially for a post-connection event, where there's often a series of specific operations you want to perform operations synchronously step-by-step to setup the peripheral and states in the app, the `backgroundTask` API is designed specifically for that.

We can also be more clear in our documentation to help understand this.

Danappelxx, 2018-08-22:
I see, my bad for not deducing that. I assumed the apis were entirely asynchronous and would resolve discrepancies like this, so this behavior seemed like a bug. I do want to say it is a bit of a confusing design as appears entirely asynchronous but really you may only safely perform one operation at a time. Using raw core bluetooth I can easily call `readValue` for all my characteristics in a loop and the delegate would process the values regardless of the order in which they come - which is kind of the point of asynchronous programming in the first place, isn't it? Forcing calls to be in sequential order seems rather limiting.

JeremyChiang, 2018-08-23:
@Danappelxx you can still mix and match asynchronous read and write calls in any order or loops without using the `backgroundTask` block, and Bluejay would actually resolve the reads and writes synchronously behind the scene for you.

The actual problem here is mixing `listen` with the reads and writes. Right now the asynchronous `listen` API doesn't play well when mixed with other asynchronous reads and writes, due to the nature of the listen operation being able to repeatedly receive future events.

This is definitely something we want to add and improve upon for Bluejay. Basically, we need to extend the read and write operations and give them ways to evaluate whether there is currently an ongoing listen, and if so, if the either the read or write operation receives an event intended for listen, it could resolve it and pass it back to the listen handler. This was too complicated and edge-case for us initially, but now might be the time.

Danappelxx, 2018-08-23:
Thank you for taking the time to answer. I think listener updates are actually fine, as they [sidestep the queue entirely](https://github.com/steamclock/bluejay/blob/master/Bluejay/Bluejay/Peripheral.swift#L412). The issue I was having was specifically enabling listens, ie [waiting for the notifications enabled delegate](https://github.com/steamclock/bluejay/blob/master/Bluejay/Bluejay/Peripheral.swift#L418) (which would only be called once). I don’t think there is a need for an edge case here - just that I believe some events should be allowed to pass down the queue.

apike, 2018-08-23:
Any thoughts on how we could make this less likely to trip people up @nbrooke?

nbrooke, 2018-08-24:
I think some of the above diagnoses from @Danappelxx and @JeremyChiang might not be correct. I don't think it's an issue of this just being a disallowed operations that needs to be messaged better, if what Dan is describing is accurate, it sounds like there is just a bug int he queuing that needs to be fixed.

The edge case that I THINK Jeremy is talking about, is that we don't (and it is very hard to) properly support interleaved listens and reads on the SAME characteristic. It's not clear to me if that is the case that Dan has from his example. If it is, maybe that's the problem, but it seems like it should be giving a very clear error in that case (it specifically checks for that).

Making a listen call immediately followed by a read call on different characteristic SHOULD be fine conceptually. When Dan says:

> listen is called, bluejay calls CB setNotifyValue(_:) method, queues ListenCharacteristic
> read is called, bluejay calls CB readValue(for:) method, queues ReadCharacteristic

That SHOULDN'T be true (though maybe that's the bug), what should happen is that `listen`should queue a `ListenCharacteristic` operation, which (assuming there aren't any other operations in flight) will start immediately and call `setNotifyValue(_:)`. However, when `read` is called right after, it should only put the `ReadCharacteristic` operation in the queue, which SHOULDN'T result in the actual CB `readValue(for:)` being called (because it's not head of the queue). Once the `peripheral(_:didUpdateNotificationStateFor:)` callback is processed then `ReadCharacteristic` should reach the front of the queue, get started, and then call `readValue(for:)`. 

If `readValue(for:)` IS getting called before the `peripheral(_:didUpdateNotificationStateFor:)` callback fires, then something is wrong with the queueing logic, that's resulting in the `ReadCharacteristic` operation starting before it should. If so, we should just need to fix that problem.

I'm not sure what could cause that, it seems like that fundamental an issue with queueing would be causing more widespread problems than this. The only thing I can think of is that it's somehow related to characteristic discovery, when you have to discover characteristics, the sequence of operations gets much more complicated. It'll be sticking in the `DiscoverCharacteristic` events and then adding the real events from the completion callbacks from those, so the two operations as a whole will interleave more. Maybe there is some case with discovering characteristics where we can trigger two operations getting started at once.

Danappelxx, 2018-08-24:
You’re right. In my example it was two separate characteristics, and after looking through bluejay another time I realize that only one operation at a time can possibly be running under the hood, so the race condition I presented is rather unlikely. I am not sure what could be causing this crash then. All I know is somehow my app crashed on an unexpected event in the read/listen characteristic operations.

nbrooke, 2018-08-27:
@Danappelxx Two questions:

- Is the case that you're seeing this also when it is reconnecting after a disconnect (like the original report)? It seems at least possible that this is some issue of stale events, maybe from previous connections, getting processed, and if so it would be a lot weirder if this was happening on initial connection than on subsequent connections. 
- Do you know what the exact operation that it is failing in and the unexpected event type that it's receiving are (first should be obvious from the callstack of the crash, the second should in theory be in the message coming from the precondition failure (something like: "Unexpected event response: Event.didDiscoverServices"), or would be easy to see if you can reproduce under the debugger.

@FIndustries If you have an answer to what the unexpected event type is, that's be useful as well (The crash log shows that it's in the DiscoverService operation failing).

Danappelxx, 2018-08-31:
Unfortunately after the crash occurred I first attempted to patch Bluejay but eventually decided to go back to my own Core Bluetooth stack, so I no longer reproduce. It may have been due to me messing with the code, or maybe the device disconnected suddenly or something else entirely. The crash occurred in  `ListenCharacteristic.process(event:)`, that’s all I can really tell you.

JeremyChiang, 2019-01-15:
v0.8.0 has addressed the problems of DiscoverService and DiscoverCharacteristic not being queued in a truly FIFO order, so this crash should be fixed now.

closed 2019-01-15 by JeremyChiang