steamclock/bluejay #162: Crash in the extract function

akmarinov, 2018-08-21:
I'm having an issue where Bluejay crashes in

[image]

Here's the call stack:

[image]

There's no console output for the crash, just that bad access.

if I do 

`        let partNumberData: Data = bluetoothData.subdata(in: 0..<0+6)
`

instead on my own, I'm able to get the sub data and use it to initialize my string to the correct value.

Any idea why you use `.withUnsafeBytes { $0.pointee }` and why it seems to crash with bad access?

Am I doing something wrong? I want to extract a string from a data that I receive.

JeremyChiang, 2018-08-21:
Thank you for finding this problem. I've filed an issue #164 with explanations and solutions for moving forward, and sorry we weren't able to catch this earlier in our development and testing.

akmarinov, 2018-08-21:
No worries, love the library!

It’s not just Strings, btw. Trying to extract a Data in this case and it still fails. (Though Strings also fail)

JeremyChiang, 2019-03-11:
At long last, we're close to having a reasonable solution for this: https://github.com/steamclock/bluejay/pull/195

I've tested the PR and both string and data extraction are now working as expected. Also, see the updated readme in that branch for more examples.