steamclock/bluejay #176: Crash when listening on multiple characteristics

beachcitiessoftware, 2018-10-26:
Summary:

I'm getting a crash when setting up several characteristic listeners.  The crash happens randomly and sometimes everything works fine, other times it crashes.

Exception Type:  EXC_BREAKPOINT (SIGTRAP)
Exception Codes: 0x0000000000000001, 0x0000000104e50330
Termination Signal: Trace/BPT trap: 5
Termination Reason: Namespace SIGNAL, Code 0x5
Terminating Process: exc handler [0]
Triggered by Thread:  0

Filtered syslog:
None found

Thread 0 name:  Dispatch queue: com.apple.main-thread
Thread 0 Crashed:
0   Bluejay                       	0x0000000104e50330 specialized ListenCharacteristic.process(event:) + 181040 (ListenCharacteristic.swift:0)
1   Bluejay                       	0x0000000104e4fc94 protocol witness for Queueable.process(event:) in conformance ListenCharacteristic + 179348 (ListenCharacteristic.swift:0)
2   Bluejay                       	0x0000000104e4fc94 protocol witness for Queueable.process(event:) in conformance ListenCharacteristic + 179348 (ListenCharacteristic.swift:0)
3   Bluejay                       	0x0000000104e5f80c specialized Queue.process(event:error:) + 243724 (Queue.swift:0)
4   Bluejay                       	0x0000000104e3feec specialized Peripheral.handleEvent(_:error:) + 114412 (Peripheral.swift:0)
5   Bluejay                       	0x0000000104e44134 specialized Peripheral.peripheral(_:didUpdateValueFor:error:) + 131380 (Peripheral.swift:0)
6   Bluejay                       	0x0000000104e3d6e0 @objc Peripheral.peripheral(_:didUpdateValueFor:error:) + 104160 (Peripheral.swift:0)
7   CoreBluetooth                 	0x0000000189c1fc18 -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 240
8   CoreBluetooth                 	0x0000000189c1fd78 -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 132
9   CoreBluetooth                 	0x0000000189c1b714 -[CBPeripheral handleMsg:args:] + 356
10  CoreBluetooth                 	0x0000000189c16350 -[CBCentralManager handleMsg:args:] + 204
11  CoreBluetooth                 	0x0000000189c24580 __30-[CBXpcConnection _handleMsg:]_block_invoke + 60

Device, Build, OS:
iPhone 8s Plus - iOS 11.4.1

JeremyChiang, 2018-11-06:
Hi @beachcitiessoftware, thank you for filing this bug. This is actually a duplicate of https://github.com/steamclock/bluejay/issues/69 and https://github.com/steamclock/bluejay/issues/141 which we will address soon after 0.7.0 is merged. 

However, the fix is not to support multiple listen on the same characteristic (at least not yet), but rather to fail gracefully and to log some warning to the console to notify the developer that you have multiple listeners - which will definitely be nicer than just crashing as it does right now!

Why do we not want to support multiple listens on the same characteristic? So far in our experiences with a variety of Bluetooth apps and frameworks, usually it's an application logic error, i.e. a programmer error, if there are multiple listeners created for one characteristic.

Bluejay believes that the best practice is to setup one listener per characteristic that needs to be listened to, and you can then have each centralized and singular listener distribute the events to the rest of your app (via your own delegates or notification systems). Of course there are always exceptions, but this is something we'd really like enforce for now.

Closing for now due to being a duplicate issue, but feel free to continue to discuss here if you have any questions!

closed 2018-11-06 by JeremyChiang

JeremyChiang, 2018-11-07:
Thanks for the clarification, let's discuss in https://github.com/steamclock/bluejay/pull/178 from now on.