steamclock/bluejay #186: backgroundTask queue, instead of stopping?

kylebrowning, 2019-01-04:
Summary: I have a use case where performing multiple background tasks is needed. Ive built a queuing system in case this happens that waits for the background task to complete, and then resumes once that is done. It would be nice if the following was possible.

- Allow multiple background tasks. (I've tested this by disabling all the isRunningBackgroundTask checks, and it doesn't seem to cause issues with our BLE Device, and writes/reads seems fine)

**OR** 

- Allow background tasks to automatically be queued.

Id be happy to submit a PR either, just curious what the reasoning behind not allowing multiple background tasks was. I presume some BLE Chips do not support multiple reads and writes at the same time?

JeremyChiang, 2019-01-04:
Hi @kylebrowning, I think I partially understand what you are looking for, but not completely. Could you kindly describe to me in details what is it you're trying to do? Perhaps there is a way to achieve what you want without needing multiple background tasks.

Based on my understanding so far, here's what I am able to answer:

The `isRunningBackgroundTask` checks exist in both the regular `read/write/listen` functions as well as in the beginning of the background task itself is to prevent other operations, whether the regular calls, or another background task, from enqueueing additional operations while a background task is already running.

A background task is supposed to guarantee the order of operations. So for example, if you have a background task that does:

`read A, write B, listen C, end listen C, write B, read A`

Bluejay uses `isRunningBackgroundTask` checks to prevent you from disturbing that order.

If you remove the `isRunningBackgroundTask` checks, and let's say `listen C` is a fairly long operation. Now somehow you also call `write D` using the regular API while `listen C` is still happening in the background task, Bluejay will then enqueue and perform `read D` after `listen C`. As a result, the order of operations now looks like:

`read A, write B, listen C, **write D**, end listen C, write B, read A`

This can be dangerous because the background task is now "betrayed" in a sense that something else happened behind the scene. For example, if `write D` does something such as turning off write permission on B, then when the background task gets to the second last operation to try to write to B, that will fail, thereby causing the background task to fail for a reason not of its fault.

kylebrowning, 2019-01-08:
Well, we have this task where we upload some bites to our BLE device. It's a user triggered action before they start to use the device. Additionally Every 30 seconds or so we grab logs from the device that are written to a FIFO stack. We read the stack, write to a characteristic about which log we just read, and then read again, giving us the next item in the stack.

Ive set both of these up as background tasks, but technically they don't interfere with each other. If one writes to a char, the other isn't affected by it. If this is the case then I think I don't need them to be background tasks, just write and read willy nilly.

JeremyChiang, 2019-01-08:
Hi @kylebrowning, thank you for your reply. Yes I believe you are right in that both the user triggered write and the repeating scheduled reads of logs should not be performed using Bluejay's background task.

The background task is meant for major, large, complex, and often one-time (per use or per session) operations where the order of execution also matters, such as upgrading a firmware, comprehensive syncing of data, or critical I/O that takes priority over all other tasks and need to be completed before other actions are allowed.

Let me know if you are still able to accomplish what you need without using the background task, and I also recommend reverting your custom changes on `isRunningBackgroundTask` to restore protection against the kinds of race-condition errors I've described earlier 👍🏻

kylebrowning, 2019-01-09:
Thanks @JeremyChiang.

closed 2019-01-09 by kylebrowning