steamclock/bluejay #188: Why this library is so big?

keyhan76, 2019-01-13:
Summary:
When I add it to my project it adds 5mb to my exported IPA file.
My IPA size is 22.6 when adding Bluejay it becomes 27.6

Steps to Reproduce:

1. 
2. 
3. 

Actual Result:

Expected Result: 

Device, Build, OS:

JeremyChiang, 2019-01-14:
Hi @keyhan76, thank you for your question. Framework size is something I'm not intimately familiar with, but after a bit of readings, here's what I can share with you in this discussion: Bluejay currently supports 4 architectures: arm64, arm64e, armv7, armv7s (for most iPhones and iPads), as well as Bitcode, which would then generate 4 slices of binaries and 4 slices of Bitcode at the end of a build. On top of that, since last Friday and the introduction of v0.8.0, Bluejay has also started uxing XCGLogger, which is a non-trivial logging framework. That could add a lot to the size as well.

Also, the full-version IPA file containing all architecture slices and bitcodes you have won't be the final product users actually download from the App Store due to app thinning (at least on iOS 9 and later). Only the relevant slice, and zero Bitcode, will be sliced from your app record and delivered to the users, resulting in a much smaller download size.

While I agree we should always strive to reduce file size, even at this age of significantly increased networking bandwidths, I think 5 ~ 10mb still seems pretty reasonable for an iOS framework for something like Bluejay. For your reference, here's another non-trivial but also not massive library and their statistics on sizes: https://docs.instabug.com/docs/sdk-size

Finally, I don't think we're doing anything unorthodox in the development process that would otherwise contribute to a larger than necessary framework size, however, if there's anything you or other people notice, please do point out and it would be a great learning experience for me as well.

To answer your question, here are the blog posts I've read:
https://pspdfkit.com/guides/ios/current/faq/framework-size/
https://www.mapbox.com/help/troubleshooting/ios-framework-size/
https://help.apple.com/xcode/mac/current/#/devbbdc5ce4f

keyhan76, 2019-01-14:
Hi @JeremyChiang,
Thanks for youe great reply.
I’m exporting this IPA under Development and Bitcode is off.
Can I remove the XCGLogger?
What’s the use of XCGLogger in the framework?

JeremyChiang, 2019-01-14:
Hello @keyhan76, I would not recommend removing XCGLogger. XCGLogger is used to allow Bluejay to log to an auto-rotating file for debugging purposes. Even though Bluejay makes things easier, Bluetooth development is still really hard to do it right. And, not having a log file to refer to when there are issues, especially in production environment, can make certain problems exponentially difficult, if not impossible, to resolve.

That said, you can still remove XCGLogger yourself if your project really calls for it. It shouldn't be too hard, but I may not be able to support you in that task if there are any problems.

keyhan76, 2019-01-14:
Ok, thanks a lot.

closed 2019-01-14 by keyhan76