steamclock/bluejay #192: Make device to act as a Peripheral

keyhan76, 2019-02-02:
Summary:
How to set the device as a peripheral so it can be discoverable by other phones or devices?
Also be able to set a service UUID for the device to make it only visible for its specific UUIDs?

JeremyChiang, 2019-02-04:
Hi @keyhan76, Bluejay doesn't support acting as a BLE peripheral. However, later this year we are going to support simulating a BLE peripheral in a new project outside of Bluejay that will serve as a companion app to Bluejay for testing.

Meanwhile, please refer to the `DittojayHeartSensorDemo` target available in the Bluejay Xcode project for examples and some answers to your questions 👍 

Also, see this particular [Apple docs](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/PerformingCommonPeripheralRoleTasks/PerformingCommonPeripheralRoleTasks.html#//apple_ref/doc/uid/TP40013257-CH4-SW1) for the most complete details and instructions. Let me know if you have any other questions, thanks!

keyhan76, 2019-02-05:
Hi @JeremyChiang thanks for your reply.
I achieved it using CoreBluetooth Framework, And I can say it's working good.

closed 2019-02-05 by keyhan76