steamclock/bluejay #196: Scan called after disconnect causes crash

jeffbk, 2019-03-11:
Summary:
If startScan is called immediately after disconnect, an exception is thrown which causes a crash.
I believe the issue arises because the Scan instance is never assigned a queue because of the isDisconnectionQueued short circuit. See https://github.com/steamclock/bluejay/blob/320dc606b7a8b61b06e5df651c3014e73a8334a7/Bluejay/Bluejay/Queue.swift#L53

This results in Scan.Fail() called, followed by Scan.stopScan(), then finally Queueable.updateQueue(), which has a guard which checks if Scan.queue is not nil. See https://github.com/steamclock/bluejay/blob/320dc606b7a8b61b06e5df651c3014e73a8334a7/Bluejay/Bluejay/Queueable.swift#L35

I think the easy fix would be to assign the queue to the Queueable before the short circuit.

I can submit a pull request for this if you would like.

However, there is an additional issue that the startScan will fail because a disconnect is queued.  It seems like this logic is misplaced as the Queue does not know wether the Queueable should fail when a disconnect is queued.

Thank you for providing such a useful library around CoreBluetooth. It simplifies many of the interactions for BLE app development.

kylebrowning, 2019-03-20:
Ive seen this crash too.

JeremyChiang, 2019-03-29:
Thanks for filing this. We've been super busy, but I will look into this next week.

kylebrowning, 2019-03-30:
@JeremyChiang I think this stems from something a bit different, most likely a race condition.

Let me see if I can put a package together.

kylebrowning, 2019-03-30:
Eh, I couldn't recreate what I thought I found, carry on.

kylebrowning, 2019-04-02:
For what it's worth @JeremyChiang I am seeing crashes on queuable as well, but they are not easy to understand.

```
0
Bluejay
$s7Bluejay15DiscoverServiceC4failyys5Error_pF + 192
1
Bluejay
$s7Bluejay5QueueC3addyyAA9Queueable_pF + 1196
2
Bluejay
$s7BluejayAACAA18PeripheralDelegateA2aCP9requested9operation4fromyAA9Operation_p_AA0B0CtFTW + 64
3
Bluejay
$s7Bluejay10PeripheralC22discoverCharactersitic33_679AD7A74107FC50D95A62D129C6175FLL_8callbackyAA24CharacteristicIdentifierV_yAA15DiscoveryResultOctFTf4xnn_n + 300
4
Bluejay
$s7Bluejay10PeripheralC4read4from10completionyAA24CharacteristicIdentifierV_yAA10ReadResultOyxGctAA10ReceivableRzlFTf4xnn_n + 1788
5
Bluejay
$s7BluejayAAC4read4from10completionyAA24CharacteristicIdentifierV_yAA10ReadResultOyxGctAA10ReceivableRzlF + 460
6
```

kylebrowning, 2019-04-02:
Okay so after further digging, this isn't just related to a scan.

1. Pair with Device.
2. Set a polling timer to request status ever 0.1 seconds
3. call `bluejay.disconnect()`
4. Observe crash.

kylebrowning, 2019-04-02:
This is some kind of race condition.

Here's a log.

```
019-04-02 14:07:32.136 [Bluejay] [main] > Should auto-reconnect: true
2019-04-02 14:07:32.136 [Bluejay] [main] > Connected to: 
[bluejaycrash.zip](https://github.com/steamclock/bluejay/files/3035938/bluejaycrash.zip)
8STZC.
2019-04-02 14:07:32.136 [Bluejay] [main] > Queue has removed Bluejay.Connection because it has finished.
2019-04-02 14:07:32.136 [Bluejay] [main] > Queue is empty, nothing to update.
2019-04-02 14:07:32.136 [Bluejay] [main] > Connection deinitialized.
2019-04-02 14:07:32.196 [Bluejay] [main] > Requesting read on Characteristic: 0000FF12-0000-1000-8000-00805F9B34FB, Service: 0000FF00-CFEA-4F81-83C0-D28E09CD44E7...
2019-04-02 14:07:32.197 [Bluejay] [main] > Queue will start Bluejay.DiscoverService...
2019-04-02 14:07:32.296 [Bluejay] [main] > Requesting read on Characteristic: 0000FF12-0000-1000-8000-00805F9B34FB, Service: 0000FF00-CFEA-4F81-83C0-D28E09CD44E7...
2019-04-02 14:07:32.396 [Bluejay] [main] > Requesting read on Characteristic: 0000FF12-0000-1000-8000-00805F9B34FB, Service: 0000FF00-CFEA-4F81-83C0-D28E09CD44E7...
2019-04-02 14:07:32.461 [Bluejay] [main] > Explicit disconnect called.
2019-04-02 14:07:32.462 [Bluejay] [main] > Bluejay.Disconnection added to queue with UUID: EEE243C1-8344-4747-96DB-D43BAF9243E9.
failure(Bluejay.BluejayError.missingService(Bluejay.ServiceIdentifier(uuid: 0000FF00-CFEA-4F81-83C0-D28E09CD44E7)))
2019-04-02 14:07:32.497 [Bluejay] [main] > Queue has removed Bluejay.DiscoverService because it has finished.
2019-04-02 14:07:32.498 [Bluejay] [main] > Queue will start Bluejay.DiscoverCharacteristic...
2019-04-02 14:07:32.499 [Bluejay] [main] > Queue has removed Bluejay.DiscoverCharacteristic because it has finished.
2019-04-02 14:07:32.499 [Bluejay] [main] > Queue will start Bluejay.DiscoverService...
2019-04-02 14:07:32.501 [Bluejay] [main] > Requesting read on Characteristic: 0000FF12-0000-1000-8000-00805F9B34FB, Service: 0000FF00-CFEA-4F81-83C0-D28E09CD44E7...
failure(Bluejay.BluejayError.disconnectQueued)
Fatal error: Containing queue is nil.: file /Users/kylebrowning/work/ios/bluejaycrash/Pods/Bluejay/Bluejay/Bluejay/Queueable.swift, line 37
2019-04-02 14:07:32.502309-0700 bluejaycrash[16338:3421008] Fatal error: Containing queue is nil.: file /Users/kylebrowning/work/ios/bluejaycrash/Pods/Bluejay/Bluejay/Bluejay/Queueable.swift, line 37
```

Attached is an example project. Obviously you will need to change the ServiceIdentifier and characteristic and possibly the `Version` class.

kylebrowning, 2019-04-02:
[bluejaycrash.zip](https://github.com/steamclock/bluejay/files/3035940/bluejaycrash.zip)

kylebrowning, 2019-04-02:
@JeremyChiang Just pinging you the above too!

JeremyChiang, 2019-04-04:
Thanks for all the info. I'll be looking at this today or tomorrow when I get a chance.

kylebrowning, 2019-04-05:
@JeremyChiang Verified. Fixed!

PR #203 (2019-04-05): Fix disconnect then scan crash
Fixes: #196 

Thanks for the great bug report @jeffbk and @kylebrowning

Failing a queueable requires updating its queue, and therefore requires the queueable to have a reference to its queue. The disconnection-is-queued failure block is relatively new, so I didn't catch the new requirement - that is to shuffle the queue assignment to the new starting point of the add call.

This should fix specifically the crashes when you make Bluejay calls right after and during a queued disconnect, such as making a scan call right after the disconnect call.

However, I am not [... cut]
closed 2019-04-05 by JeremyChiang