steamclock/bluejay #197: Accessory that request a password at connect

soufianeEssabbane, 2019-03-13:
Summary:

Hi, while trying to connect to an accessory that requires a password, a popup shows and dismiss rapidly, without even having time to write the password. how can we handle this kind of connexions ?
like making at least the popup wait the time needed to write the password ? or how can we write the password programatically during bluejay.connect ? 

Steps to Reproduce:

1. scan for accessory 
2. selected accessory 
3. make bluejay.connect ...

Actual Result:
a natif popup for password displays and dismiss rapidly.

Expected Result: 
send accessory password programmatically or make natif popup password wait

Device, Build, OS:
iOS

JeremyChiang, 2019-03-29:
Hi @soufianeEssabbane, it sounds like you have some UI bugs unrelated to Bluejay. There's no code in Bluejay that can show or dismiss an alert controller. I would suggest finding code that dismisses (or could potentially dismiss) alerts your app and trace back from there to see what is causing the alert to disappear unexpectedly.

JeremyChiang, 2019-04-04:
Hi @soufianeEssabbane did you make any progress on resolving this issue? Let me know. Otherwise, I will be closing this issue soon for inactivity. Thanks.

closed 2019-06-27 by JeremyChiang

soufianeEssabbane, 2019-08-08:
Hi @JeremyChiang sometimes for some BLE devices some characteristics are protected with a password, when you try to subscribe to notification from a characteristic or read a protected one, the OS expose to you a native Popup for password.
but when you read or subscribe to many protected characteristics you should wait for the user to write the password that popups for the first charateristic before continuing reading the rest ... otherwise the popup will be dismissed.