steamclock/bluejay #198: Crash report: Bluejay.swift: 1437

kylebrowning, 2019-03-20:
Summary: Forgetting a device in iOS settings, while attempting to connect causes a crash.

Steps to Reproduce:

1. Request a get or write an encrypted attribute to get a pairing
2. Turn off device so that it disconnects unexpectedly and Bluejay attempts to reconnect
3. Forget device in iOS BLE Settings

Actual Result: App crashes on line 1437 in Bluejay.swift

```swift
guard let connectingError = connectingError else {
                        preconditionFailure("Missing connecting error at the end of a disconnect clean up after cancelling a pending connection.")
                    }
```

Expected Result: 
App should not crash

Device, Build, OS:
iPhone XS Max, iOS 12.1, Bluejay (0.8.1) `b01d53f0592cc1cf26382d8df1734c16e83286e4`

JeremyChiang, 2019-03-29:
Also thanks for filing this. I need to look into this more, but initial investigation suggests that we might need to figure what CoreBluetooth does when a device is unpaired, and assign the correct error to `connectingError` before this check occurs.

FIndustries, 2019-04-02:
I have the same problem just disabling the bluetooth from the control center. It crashes a lot of times but not every time.

apike, 2019-04-02:
Let's try and get a fix for this into a point release, rather than holding to the next scheduled update.

kylebrowning, 2019-04-02:
@apike It'd be nice to get this in too, https://github.com/steamclock/bluejay/pull/200

But I'd understand fi that needs a more scheduled update.

PR #202 (2019-04-03): v0.8.3 - Fix unpairing crash
Fixes: #198 

@kylebrowning could you also review this PR? Thanks 👍🏻

## Summary
This fix allows Bluejay to complete the disconnection clean up without crashing when a device is unpaired.

## Solution
I've never triggered iOS' system pairing dialog before with Bluejay, because I've never worked with a characteristic that is encrypted, since most of our client projects deal with authentication upfront, not on per-characteristic basis.

But I agree this can be common, so this fix is important.

Anyhow, I didn't know we can also get a disconnection event while the con [... cut]
closed 2019-04-03 by JeremyChiang