steamclock/bluejay #205: Multiple random crashes in queue.swift

kylebrowning, 2019-04-19:
Honestly, I havn't been able to track down why these are happening. My gut is telling me it's related to making a request while my BLE device is turning itself off. I completely understand if this ticket gets closed but Im really at a loss as to why these are happening and am not able to reproduce. I wanted to create a ticket for posterity.

[image]

The left number is the number of crashes, and the right number is the unique user count.\

FWIW, my example code with the BLECrash in the previous tickets suffers from this as well, but I cant reliably reproduce and at a loss :(

One crash appears to happening if my BLEDevice automatically turns off, and I attempt to reconnect.
And that happens here:
```swift
        if isCBCentralManagerReady {
            precondition(queue.isEmpty, "Queue is active and is not emptied at the end of cancel all.")
```

SaintNicholas, 2019-06-19:
I've seen a lot of crashes for that same thing. I pasted the Bluejay log below from when it happened. At the time of the crash, the app had started 3 simultaneous writes, and then very shortly after it did a disconnect with immediate set to true.

It appears that what's happening is that either:
- The immediate disconnect isn't actually removing everything from the queue.
- Between the time that the immediate disconnect removes everything from the queue and the time that the queue is checked to be sure that it's empty, something else is added to the queue. There doesn't appear to be anything in the bluejay library to lock the queue while it's being modified in the cancelAll function.

// Connected to the peripheral.
2019-06-19 15:17:12.523 [Bluejay] [main] > Bluejay.Connection added to queue with UUID: 7A227EC1-3A86-4785-BE22-4318A6C272B1.
2019-06-19 15:17:12.524 [Bluejay] [main] > Init Peripheral: <Bluejay.Peripheral: 0x2812f0d20>, Optional(<CBPeripheral: 0x282da8320, identifier = DC02558C-CF8E-EC0A-A177-D3F819407226, name = Mirror, state = disconnected>)
2019-06-19 15:17:12.524 [Bluejay] [main] > Queue will start Bluejay.Connection...
2019-06-19 15:17:12.525 [Bluejay] [main] > Started connecting to HomeLink Mirror.
2019-06-19 15:17:12.861 [Bluejay] [main] > Central manager did connect to: HomeLink Mirror
2019-06-19 15:17:12.861 [Bluejay] [main] > Should auto-reconnect: true
2019-06-19 15:17:12.862 [Bluejay] [main] > Connected to: HomeLink Mirror.
2019-06-19 15:17:12.862 [Bluejay] [main] > Queue has removed Bluejay.Connection because it has finished.
2019-06-19 15:17:12.862 [Bluejay] [main] > Queue is empty, nothing to update.
2019-06-19 15:17:12.863 [Bluejay] [main] > Connection deinitialized.

// 3 parallel Bluetooth writes are started. 
2019-06-19 15:17:13.075 [Bluejay] [main] > Requesting write on Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4...
2019-06-19 15:17:13.076 [Bluejay] [main] > Queue will start Bluejay.DiscoverService...
2019-06-19 15:17:13.076 [Bluejay] [main] > Queue has removed Bluejay.DiscoverService because it has finished.
2019-06-19 15:17:13.076 [Bluejay] [main] > Queue is empty, nothing to update.
2019-06-19 15:17:13.077 [Bluejay] [main] > Queue will start Bluejay.DiscoverCharacteristic...
2019-06-19 15:17:13.078 [Bluejay] [main] > Requesting write on Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4...
2019-06-19 15:17:13.080 [Bluejay] [main] > Bluejay.WriteCharacteristic<App.Sendable> added to queue with UUID: 7A227EC1-3A86-4785-BE22-4318A6C272B1.
2019-06-19 15:17:13.081 [Bluejay] [main] > Queue has removed Bluejay.DiscoverCharacteristic because it has finished.
2019-06-19 15:17:13.081 [Bluejay] [main] > Queue will start Bluejay.DiscoverService...
2019-06-19 15:17:13.081 [Bluejay] [main] > Queue has removed Bluejay.DiscoverService because it has finished.
2019-06-19 15:17:13.081 [Bluejay] [main] > Queue will start Bluejay.DiscoverCharacteristic...
2019-06-19 15:17:13.081 [Bluejay] [main] > Bluejay.WriteCharacteristic<App.Sendable> added to queue with UUID: 7A227EC1-3A86-4785-BE22-4318A6C272B1.
2019-06-19 15:17:13.082 [Bluejay] [main] > Queue has removed Bluejay.DiscoverCharacteristic because it has finished.
2019-06-19 15:17:13.082 [Bluejay] [main] > Queue will start Bluejay.WriteCharacteristic<App.Sendable>...
2019-06-19 15:17:13.083 [Bluejay] [main] > Started write to Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4 on DC02558C-CF8E-EC0A-A177-D3F819407226.
2019-06-19 15:17:13.156 [Bluejay] [main] > Write to Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4 on DC02558C-CF8E-EC0A-A177-D3F819407226 is successful.
2019-06-19 15:17:13.157 [Bluejay] [main] > Queue has removed Bluejay.WriteCharacteristic<App.Sendable> because it has finished.
2019-06-19 15:17:13.158 [Bluejay] [main] > Queue will start Bluejay.WriteCharacteristic<App.Sendable>...
2019-06-19 15:17:13.160 [Bluejay] [main] > Started write to Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4 on DC02558C-CF8E-EC0A-A177-D3F819407226.
2019-06-19 15:17:13.167 [Bluejay] [main] > Requesting write on Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4...
2019-06-19 15:17:13.176 [Bluejay] [main] > Explicit disconnect called.
2019-06-19 15:17:13.176 [Bluejay] [main] > Cancel everything called with error: Explicit disconnect, shouldDisconnect: true
2019-06-19 15:17:13.176 [Bluejay] [main] > Queue will now cancel all operations with error: Explicit disconnect
2019-06-19 15:17:13.176 [Bluejay] [main] > Failed writing to Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4 on DC02558C-CF8E-EC0A-A177-D3F819407226 with error: Explicit disconnect
2019-06-19 15:17:13.176 [Bluejay] [main] > Queue has removed Bluejay.WriteCharacteristic<App.Sendable> because it has finished.
2019-06-19 15:17:13.176 [Bluejay] [main] > Queue will start Bluejay.DiscoverService...
2019-06-19 15:17:13.177 [Bluejay] [main] > Queue has removed Bluejay.DiscoverService because it has finished.
2019-06-19 15:17:13.177 [Bluejay] [main] > Queue will start Bluejay.DiscoverCharacteristic...
2019-06-19 15:17:13.177 [Bluejay] [main] > Bluejay.WriteCharacteristic<App.Sendable> added to queue with UUID: 7A227EC1-3A86-4785-BE22-4318A6C272B1.
2019-06-19 15:17:13.177 [Bluejay] [main] > Queue has removed Bluejay.DiscoverCharacteristic because it has finished.
2019-06-19 15:17:13.177 [Bluejay] [main] > Queue will start Bluejay.WriteCharacteristic<App.Sendable>...
2019-06-19 15:17:13.177 [Bluejay] [main] > Started write to Characteristic: 5323D9D9-7B50-4EC2-B2D5-E94DE52668F3, Service: 9D8E58B2-1D2D-4DC4-B324-EDDF69746AD4 on DC02558C-CF8E-EC0A-A177-D3F819407226.
<Crash>

JeremyChiang, 2019-06-20:
Thank you for reporting this. This is on my radar to look at next by early next week. Please check in again with me if you don't hear back from me.

SaintNicholas, 2019-07-11:
Hey @JeremyChiang, just checking in. Thanks.

JeremyChiang, 2019-07-12:
Hi @kylebrowning and @SaintNicholas sorry for the delay, we've just been super busy with client work, but I want to help.

It seems like there are 2 different crashes reported here.

1. Cancel all called but queue is not emptied by the end
2. An extra read request got snuck into the Bluejay queue after a call to disconnect

For Kyle's crash, it may be useful to log a runtime error to your analytics/crash report service:
```
if isCBCentralManagerReady {
    // [insert call to log maybe the entire state of the queue, or the first item still in the queue]
    precondition(queue.isEmpty, "Queue is active and is not emptied at the end of cancel all.")
} else {
    precondition(!queue.contains { queueable -> Bool in
        !queueable.state.isFinished
    }, "Queue is inactive but still contains unfinished queueable(s) at the end of cancel all.")
}
```
Figuring out what is left in the queue can help us backtrace and figure out why the queue isn't emptied as expected.

For Saint Nicholas' crash, I suspect somewhere in your app or callbacks you might be issuing an extra read, after another part of your app or callback has started a disconnect process. See the highlighted section from my screenshot of your logs. If my suspicion is correct, you might be nesting a read request in another callback, where your nested read request fires after its parent callback got cancelled because a disconnect.

There is a Bluejay bug here in that we should protect you better by failing your nested callbacks for you if Bluejay is disconnecting or shutting down, but you can also handle that yourself for now by either avoid nesting callbacks (use the Bluejay background task instead), or by checking for Bluejay's states such as `isDisconnecting` and `isConnecting` and `isConnected` inside your callbacks before you issue further Bluejay request in them.

This seems like a pretty easy and important fix. I will plan to roll out an official fix for this specific case.

elprl, 2020-04-22:
Any update on this? I'm still seeing this issue.

linxiaoyuan, 2024-08-06:
Still keep waiting for the solution since 2019

dorsafguesmi-eng, 2025-05-12:
Any solution for this problem ?