steamclock/bluejay #210: Crash in disconnection cleanup

kylebrowning, 2019-06-17:
Summary: On average we get about 60 crashes a week with on the following line.
```swift
            if wasConnecting || weakSelf.queue.isRunningQueuedDisconnection {
                precondition(
                    !weakSelf.queue.isEmpty,
                    "Queue should not be emptied at the beginning of disconnect clean up when Bluejay was still connecting or has started a queued disconnection."
                )
```

Steps to Reproduce:

Unknown.

Since I can't find a way to reproduce this, im at the mercy of steamclock here.

Any ideas on what might cause this to happen.

JeremyChiang, 2019-06-17:
Are you able to find out whether Bluejay was still connecting, or whether there was a queued disconnect operation?

This precondition is to make sure two things:

1) If you are disconnected while still connecting, there has to be a pending connection operation in the queue. Something is wrong if you are disconnected while still connecting, but if there is no connection operation to cancel in the queue.
2) If you are disconnected during a queued disconnection call, like above, there needs to be a disconnect operation in the queue to cancel, or else the queue has also got into an unexpected state.

It's possible you have encountered an additional edge case or bug that we don't handle properly yet, but for now we should check to see if there isn't any unexpected Bluetooth operation in your app that is triggering this precondition, as it is originally designed to call out something wrong and a corrupted queue state.

kylebrowning, 2019-06-17:
@JeremyChiang Thanks for the clarifications. Ill see if I can pinpoint it.

I call disconnect if there are any pairing failures. I did this because if I didn't I get the error that multiple connect is not supported.

Is there a better way to handle that scenario?

JeremyChiang, 2019-06-17:
Note that there are 2 versions of disconnect, an immediate one, and a queued one.

The immediate disconnect would cancel everything in the queue right off the bat and start disconnecting as soon as the queue is completely emptied with a disconnection event propagated to all of the items previously in the queue.

The queued disconnect would insert an operation at the end of the queue, and wait for its turn before actually performing a disconnect.

Which one you should use is up to you and your app's needs, but it's good to be aware of the differences and choose the right tool for the job.

With that context out of the way, it seems like you are using the queued disconnect call. Is it possible in your pairing failures' callbacks, you are also disconnecting or doing something that would empty the queue? I can see this crash happening if that is the case - a case of redundant disconnection and failure handling.

kylebrowning, 2019-06-17:
Thanks for this insight, I've adjusted how were pair slightly in the failure conditions and will report back on crash rates.

closed 2019-06-17 by kylebrowning