steamclock/bluejay #211: No disconnect callback when Bluetooth Powered Off

SaintNicholas, 2019-06-21:
Summary: When connected to a peripheral, and I turn off Bluetooth, I don't think the proper things are happening. There are several questions I have.

- I get a message of API MISUSE: <CBCentralManager: 0x115480c60> can only accept this command while in the powered on state
- I never get the ConnectionObserver.disconnected function call, and I had expected that I would, since the bluejay logs say that it is disconnecting, and my CBPeripheral.state value indicates that it disconnected?

Steps to Reproduce:

1. Be connected to a peripheral.
2. Turn off Bluetooth.

Actual Result:

2019-06-21 13:59:37.192 [Bluejay] [main] > Central manager state updated: Powered Off
2019-06-21 13:59:37.192 [Bluejay] [main] > Cancel everything called with error: Bluetooth unavailable, shouldDisconnect: true
2019-06-21 13:59:37.193 [Bluejay] [main] > Queue is empty, nothing to cancel.
2019-06-21 13:59:37.193 [Bluejay] [main] > Cancel everything will now disconnect a connected peripheral...
2019-06-21 13:59:37.193 [Bluejay] [main] > Should auto-reconnect: false
2019-06-21 13:59:37.193477-0400 App[1832:416758] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1169796a0> can only accept this command while in the powered on state
2019-06-21 13:59:37.194 [Bluejay] [main] > Cleared all restoration peripheral references.
2019-06-21 13:59:37.194 [Bluejay] [main] > Deinit Peripheral: <Bluejay.Peripheral: 0x117857890>, Optional(<CBPeripheral: 0x10e932b60, identifier = DC02558C-CF8E-EC0A-A177-D3F819407226, name = Peripheral, state = disconnected>))
2019-06-21 13:59:37.195 [Bluejay] [main] > Cleared all non restoration peripheral references.

Expected Result: 

I get the disconnected callback?

Device, Build, OS:

iPhone Xs, Bluejay 0.8.5, iPhone 12.3.1

JeremyChiang, 2019-06-26:
This also a very good question. Thanks for asking.

The TLDR is disconnect and Bluetooth off in terms of CoreBluetooth might need to be treated as 2 distinct and unrelated events.

There is something I would like to improve here in Bluejay, but even after cleaning things up, while the logs and the sequence of events will be less confusing, you would still not get a disconnect callback when Bluetooth is turned off.

That's because the disconnect callbacks are reserved for actual disconnect events that took place (whether CB call `didDisconnect`), and when Bluetooth is turned off, I believe CoreBluetooth does not actually call `didDisconnect` at all. Though this could change in any future iOS update.

At least for now, and in the future, you might need to handle disconnection and Bluetooth off separately, because that's also how CoreBluetooth deals with these 2 events. I also see the possibility of Bluejay just simply notifying the disconnect listeners on behalf of CoreBluetooth when Bluetooth is turned off, but that needs a bit more thought and experimentation.

So on that API MISUSE log you saw:
-  API MISUSE: <CBCentralManager: 0x115480c60> can only accept this command while in the powered on state
    - This is because we try to call disconnect when the queue is emptied, which doesn't do anything when Bluetooth is turned off (something Bluejay can clean up)

SaintNicholas, 2019-07-03:
Thanks @JeremyChiang. I am now handling the Bluetooth off and the disconnect as separate events.

JeremyChiang, 2019-07-11:
Great, closing this for now.

closed 2019-07-11 by JeremyChiang

mudassar1122, 2022-06-20:
When i was connected to a peripheral; some seconds device on and i have found this error.
"The connection has timed out unexpectedly."