steamclock/bluejay #217: Crash reading a characteristic: "Expecting read from \(characteristicIdentifier.description), but actually read from \(readFrom.uuid)"

SaintNicholas, 2019-07-10:
Summary:

We haven't been able to reproduce it, because we're not sure when/how it occurs, but our logs show several crashes on ReadCharacteristic.swift line 57, which says "Expecting read from \(characteristicIdentifier.description), but actually read from \(readFrom.uuid)".

Steps to Reproduce:

Not sure.

Actual Result:

Crash.

Expected Result: 

Doesn't crash?

Device, Build, OS:

iPhone 6s, iOS 12.3.1, Bluejay 0.8.5.

JeremyChiang, 2019-07-11:
Hmmm...this is also not enough info for me to provide a useful answer, but I do have 2 suggestions:

1. Investigate whether your app is causing Bluejay to make CB issue more than 1 read request. Bluejay is designed so that it can only work with 1 read or write at a time. You can issue multiple read or write calls asynchronously, but Bluejay will queue them synchronously in a FIFO order. If you somehow get CB to issue more than 1 read request, then while Bluejay is expecting a read result from A, the read result from B could come back first instead, causing this precondition - which indicates the Bluejay queue is corrupted and something has gone terribly wrong.
2. This could also be a Bluetooth LE peripheral problem. If your peripheral is not responding to read requests properly, this could happen. For example, you issued a read request on characteristic A, but your Bluetooth LE firmware decides to send a read response through characteristic B.

SaintNicholas, 2019-07-11:
@JeremyChiang

I will investigate more, but I think all of our reads/writes are going through Bluejay, so we shouldn't be able to issue more than 1 write at a time to Bluejay.

We are receiving notifications as well as doing explicit reads. I think, in iOS, the results for both of these come through the `optional func peripheral(_ peripheral: CBPeripheral, didUpdateValueFor characteristic: CBCharacteristic, error: Error?)` function. Is it possible that after an explicit read is requested, and before that read value is returned, a notification value, from a different characteristic, is returned, and that's what's causing this crash?

JeremyChiang, 2019-07-11:
Ah, yes, very good point 👍 You may need to restore listens! https://github.com/steamclock/bluejay#listen-restoration

SaintNicholas, 2019-07-13:
Thanks @JeremyChiang, I'll try it next week and let you know how it goes.

JeremyChiang, 2019-10-01:
Hi @SaintNicholas just checking in to see if you have resolved this crash, thanks.

JeremyChiang, 2019-10-17:
Closing for inactivity.

closed 2019-10-17 by JeremyChiang

Maikol, 2020-10-29:
Hi @JeremyChiang sorry for reusing an old issue but we're having the same problem so figured I'd comment here.

We are listening to a characteristic and while this is going on we do reads on different characteristics as well. Occasionally we get crashes on `ReadCharacteristic.swift:57`

```swift
preconditionFailure("Expecting read from \(characteristicIdentifier.description),
    but actually read from \(readFrom.uuid)")
```

where the read from is the characteristic we're listening too. 

Are we expected to stop all listeners before performing a read? Might be more noticeable in our case since broadcasting occurs every 1 second. 

Thank you!