steamclock/bluejay #233: Why does scanning get cancelled before it's had a chance to report any discovery

sarbogast, 2020-03-04:
My goal is to scan for devices with a set of known services, get a location for the device, and send all that information to the backend, in the foreground or in the background, either when a new peripheral gets scanned or when a signification location change is detected. I'm not connecting to the peripherals or their services, I'm only using the peripheral-specific identifier advertised in the advertisement packet.

For now, I'm testing the manual scan I can trigger from a button in my user interface, that calls the `quickFrontScan()` function below:

```
    func quickFrontScan() {
        if UIApplication.shared.applicationState != .active {
            return
        }
        log.debug("Performing manual scan in the foreground. Locating user...")
        LocationManager.shared.locateFromGPS(.oneShot, accuracy: .block) { [weak self] result in
            switch result {
            case .failure(let error):
                log.error("Error locating user: \(String(describing: error))", context: "AppDelegate.quickFrontScan()")
            case .success(let location):
                if let weakSelf = self {
                    log.debug("User located in [\(location.coordinate.latitude), \(location.coordinate.longitude)]", context: "AppDelegate.quickFrontScan()")
                    weakSelf.scanForLocation(location)
                }
            }
        }
    }

    func scanForLocation(_ location: CLLocation) {
        log.debug("Scanning for bluetooth tags for location [\(location.coordinate.latitude),\(location.coordinate.longitude)]...", context: "AppDelegate.scanForLocation(_:)")

        let scanner = Bluejay()
        scanner.start()
        scanner.scan(
                duration: ScanDurationOnLocationChange,
                allowDuplicates: true,
                throttleRSSIDelta: 5,
                serviceIdentifiers: self.scannedServices,
                discovery: { [weak self] (discovery, discoveries) -> ScanAction in
                    guard let _ = self else {
                        return .stop
                    }
                    log.debug("Discovered \(discovery.peripheralIdentifier.description) with advertisement packet \(discovery.advertisementPacket) and RSSI \(discovery.rssi)")
                    log.debug("\(discoveries.count) devices discovered so far.")
                    return .continue
                }, expired: { (discovery, discoveries) -> ScanAction in
            log.debug("\(discovery.peripheralIdentifier.description) disappeared")
            log.debug("\(discoveries.count) devices remaining.")
            return .continue
        }) { [weak self] (discoveries, error) in
            log.debug("Scan completed for location [\(location.coordinate.latitude),\(location.coordinate.longitude)]", context: "AppDelegate.scanForLocation(_:)")
            if let error = error {
                log.error("An error occurred while scanning on location: \(error.localizedDescription)", context: "AppDelegate.configureLocation()")
            } else {
                self?.sendTagSeenReport(location: location, tags: discoveries)
            }
        }
    }
```

Here is the logs I get after I call the `quickFrontScan()` function:

> 20:09:17 DEBUG Performing manual scan in the foreground. Locating user...
> 20:09:17 DEBUG Significant location change recorded. New location: (-39.19141876880518, 173.93969214548974)
> 20:09:17 DEBUG Scanning for bluetooth tags for location [-39.19141876880518,173.93969214548974]...
> 2020-03-04 20:09:17.831 [Bluejay] [main] > Bluejay initialized with UUID: 632F3185-D33B-48E8-8745-9901D0411BC7.
> 2020-03-04 20:09:17.831 [Bluejay] [main] > Queue initialized with UUID: 0104CF71-4DA6-4796-AB43-19F0A99499C0.
> 2020-03-04 20:09:17.831734+1300 HeronTrack[6409:3846648] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x281007020> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
> 2020-03-04 20:09:17.833 [Bluejay] [main] > CBCentralManager initialized.
> 2020-03-04 20:09:17.833 [Bluejay] [main] > Bluejay with UUID: 632F3185-D33B-48E8-8745-9901D0411BC7 started.
> 2020-03-04 20:09:17.833 [Bluejay] [main] > Bluejay.Scan added to queue with UUID: 0104CF71-4DA6-4796-AB43-19F0A99499C0.
> 2020-03-04 20:09:17.833 [Bluejay] [main] > Queue is paused because CBCentralManager is not ready yet.
> 2020-03-04 20:09:17.833 [Bluejay] [main] > Cancel everything called with error: Cancelled, shouldDisconnect: true
> 2020-03-04 20:09:17.833 [Bluejay] [main] > Queue will now cancel all operations with error: Cancelled
> 2020-03-04 20:09:17.833 [Bluejay] [main] > Stop scan requested with error: Cancelled
[... 36 lines cut ...]
> 2020-03-04 20:09:17.852 [Bluejay] [main] > Queue is paused because CBCentralManager is not ready yet.
> 2020-03-04 20:09:17.852 [Bluejay] [main] > Deinit Bluejay with UUID: E2FC68D4-3F44-4763-8633-623CB744A4B8.
> 2020-03-04 20:09:17.854 [Bluejay] [main] > Bluejay initialized with UUID: 68236E7E-280A-4437-94C1-385F8F97499B.
> 2020-03-04 20:09:17.854 [Bluejay] [main] > Queue initialized with UUID: EECCE750-4712-477A-BA1E-6EC072CCAC74.
> 2020-03-04 20:09:17.853988+1300 HeronTrack[6409:3846648] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x2810560d0> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
> 20:09:17 DEBUG Scan completed for location [-39.19141876880518,173.93969214548974]
> 2020-03-04 20:09:17.854 [Bluejay] [main] > CBCentralManager initialized.
> 20:09:17 ERROR An error occurred while scanning on location: Cancelled
> 2020-03-04 20:09:17.854 [Bluejay] [main] > Bluejay with UUID: 68236E7E-280A-4437-94C1-385F8F97499B started.
> 20:09:17 DEBUG Significant location change recorded. New location: (-39.19141876880518, 173.93969214548974)
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Bluejay.Scan added to queue with UUID: EECCE750-4712-477A-BA1E-6EC072CCAC74.
> 20:09:17 DEBUG Scanning for bluetooth tags for location [-39.19141876880518,173.93969214548974]...
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Queue is paused because CBCentralManager is not ready yet.
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Cancel everything called with error: Cancelled, shouldDisconnect: true
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Queue will now cancel all operations with error: Cancelled
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Stop scan requested with error: Cancelled
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Scanning stopped with error: Cancelled
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Queue is paused because CBCentralManager is not ready yet.
> 2020-03-04 20:09:17.855 [Bluejay] [main] > Deinit Bluejay with UUID: 68236E7E-280A-4437-94C1-385F8F97499B.
> 20:09:17 DEBUG Scan completed for location [-39.19141876880518,173.93969214548974]
> 20:09:17 ERROR An error occurred while scanning on location: Cancelled

Note that I am using [SwiftLocation](https://github.com/malcommac/SwiftLocation) for all my location requests.

What I don't understand is the "Queue will now cancel all operations with error: Cancelled" part. What requested my scan to be cancelled and jump straight to the end part of my scan, without even waiting for the timeout (ScanDurationOnLocationChange = 10.0).

JeremyChiang, 2020-03-05:
Hi @sarbogast, your Bluejay instance is only scoped to the function, so as soon as the function `scanForLocation` exits, your Bluejay instance is also getting deallocated as indicated by the multiple `deinit` calls in the logs.

I think you can fix the issue by having your view controller hold onto the Bluejay instance via making it a property (instance variable) of the view controller.

For more information, you can also try googling "instance variable vs local variable lifecycle" Hope that helps 👍🏻

closed 2020-03-05 by sarbogast

sarbogast, 2020-03-05:
Indeed, that solved the issue. Thanks. Might be interesting to add this warning to the documentation, because I was not used to this sort of scenario.