steamclock/bluejay #259: Add support for iOS 15 / Xcode 13

LordParsley, 2021-09-30:
Summary:

When using Xcode 13 to build an iOS 15 app with the Bluejay dependency, there are compiler errors:

```
Value of optional type 'CBService?' must be unwrapped to refer to member 'uuid' of wrapped base type 'CBService'
```

Steps to Reproduce:

1. Include `pod 'Bluejay'` as a Cocoapods dependency.
2. Open workspace in Xcode 13.0.
3. Try build or run the project.

Actual Result:

[image]

Expected Result: 

Bluejay supports the newer CoreBluetooth API which has an optional `CBService` property on `CBCharacteristic` and the build succeeds.

```
    /**
     * @property service
     *
     *  @discussion
     *      A back-pointer to the service this characteristic belongs to.
     *
     */
    weak open var service: CBService? { get }
```

Device, Build, OS:

Any simulator or device building against iOS 15.

LordParsley, 2021-09-30:
Thanks for the great library! Any attention on this would be much appreciated! 😊 

FYI there is a PR submitted which checks the compiler version & then force-unwraps the service: https://github.com/steamclock/bluejay/pull/258.

It might be safer to change the `CharacteristicIdentifier` `service` property to an optional here https://github.com/steamclock/bluejay/blob/master/Bluejay/Bluejay/CharacteristicIdentifier.swift#L15 but that causes a lot of downstream changes where `.service` is being accessed.

A default service value could be used [but that might be bad practice because I'm not sure whether a zero UUID is reserved or even valid]:

```
    /// Create a `CharacteristicIdentifier` using a `CBCharacterstic`.
    public init(_ cbCharacteristic: CBCharacteristic) {
        self.service = ServiceIdentifier(uuid: cbCharacteristic.service?.uuid ?? CBUUID(string: "00000000-0000-0000-0000-000000000000"))
        self.uuid = cbCharacteristic.uuid
    }
```

Any discussion welcome!

OmarSpyras, 2021-11-16:
Hi any news on adding support ?

shamangeorge, 2021-11-18:
What is the status on this?

nbrooke, 2021-11-19:
Opened a PR that fix this (#260), should be merged and released in at most a few days, but if anyone wants to try it out faster, you can pull from that branch in your package manager of choice.

nbrooke, 2021-11-24:
Fixed in v0.8.9

closed 2021-11-24 by nbrooke